AdamRJensen/SARC #12: Variable types for saving

AdamRJensen, 2025-03-18:
To save space, we should consider how many decimals we include in the final data files for each type of parameter.

One way to control this is to set the "data type" when writing data to a table (exporting).

For most variables, the FP2 type should be sufficient, whereas for Irradiance it would be nice to always have at least one decimal point (which would require IEEE4).

[image]

AdamRJensen, 2025-03-18:
List of non-FP2 variables:
- Irradiances
- Air pressure
- Solar position
- Relative humidity
- Fan speed

AdamRJensen, 2025-03-24:
The solys azimuth and tilt has quite some decimals: 89.7167,90.3579

Keeping two decimals probably makes more sense.

Can we test if we round to two decimals if then the decimals don't match?

AdamRJensen, 2025-04-02:
Let's say you save the number 198.2, does it really matter if it is saved as an FP2 or IEEE4? Would IEEE4 add non-existing trailing zeros?

IoannisSifnaios, 2025-04-03:
> * [ ]  Let's say you save the number 198.2, does it really matter if it is saved as an FP2 or IEEE4? Would IEEE4 add non-existing trailing zeros?
> 
> I'm posting this as it would simplify things significantly if we could just output everything as IEEE4. Both save lines of code but also save potential errors as that in [#32](https://github.com/AdamRJensen/SARC/issues/32)

I think you are correct. If the value is 198.2 it would not matter if you save it as FP2 or IEE4.

The "problem" I see is that there are some values that have many decimals that we might not care to store all of them (e.g., voltage output or the tracker position). If we agree to round these to some decimal that makes sense for us, then I guess we could save everything as IEE4 (this would also simplify a lot the Table structure)

AdamRJensen, 2025-04-03:
> The "problem" I see is that there are some values that have many decimals that we might not care to store all of them (e.g., voltage output or the tracker position). If we agree to round these to [...]

But all the parameters with lots of decimals are probably currently being saved as IEEE4, whereas the parameters saved as FP2 are probably mainly values with only one or two decimals. @IoannisSifnaios Do you agree?

IoannisSifnaios, 2025-04-03:
> But all the parameters with lots of decimals are probably currently being saved as IEEE4, whereas the parameters saved as FP2 are probably mainly values with only one or two decimals. [...]

Not quite. From a quick search I see that the following parameters have many decimals but are currently saved as FP2:

- Tilt angles (MS80 and SR300)
- Pitch and Roll (SMP12)
- Rotation (SR300)
- Relative humidity (MS80, SR300, SRD100)
- Temperature (SRD100, MS80, CHP1, CGR4)

These parameters should be rounded to an agreed decimal point before saved as IEE4

AdamRJensen, 2025-04-05:
At 01:25 local time on 06-04-2025 I changed FP2 to IEEE4 for the secondary.

IoannisSifnaios, 2025-05-02:
From 02-05-2025 onwards, all the parameters on the primary are saved as IEEE4

closed 2025-05-02 by IoannisSifnaios