AdamRJensen/SARC #49: Output nan values when modbus value are integers

AdamRJensen, 2025-04-09:
In crbasic, integers can't be nan. If they are set to nan they are simply set to the lowest possible number.

Therefore, the below logic should be improved such that the final converted float data (MS80SHplus_1209_data) is NAN isf the result code fails.

```
ModbusClient(MS80SHplus_1209_result_code,ComC3,19200,MS80SHplus_1209_modbus_addr,3,MS80SHplus_1209_raw_data_unsigned,21+1,1,MODBUS_RETRIES,MODBUS_TIMEOUT,3)
    If MS80SHplus_1209_result_code <> 0 Then MS80SHplus_1209_raw_data_unsigned = NAN
    MS80SHplus_1209_data(10) = MS80SHplus_1209_raw_data_unsigned / 10
    ''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''
```

which can be achieved by switching the last two lines and updating the variable which should be nan.
```
ModbusClient(MS80SHplus_1209_result_code,ComC3,19200,MS80SHplus_1209_modbus_addr,3,MS80SHplus_1209_raw_data_unsigned,21+1,1,MODBUS_RETRIES,MODBUS_TIMEOUT,3)
    MS80SHplus_1209_data(10) = MS80SHplus_1209_raw_data_unsigned / 10
    If MS80SHplus_1209_result_code <> 0 Then MS80SHplus_1209_data(10) = NAN
    ''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''
```

AdamRJensen, 2025-05-01:
This has been done for the seconary.

IoannisSifnaios, 2025-05-02:
Also done for the primary.