AdamRJensen/SARC #54: Issue with secondary loggers slow sequence

AdamRJensen, 2025-04-16:
[image]

[image]

AdamRJensen, 2025-04-30:
The logging of some of these parameters is not on a consistent frequency

[image]

IoannisSifnaios, 2025-05-01:
The measurements of most instruments in the secondary slow sequence are not continuously logged. This is probably due to the datalogger running out of time halfway through the slow scan. However, these measurements change very slowly (e.g., tilt, internal relative humidity, heater current, fan speed), so it probably can remain as is.

This is also described here #56

IoannisSifnaios, 2025-05-02:
@AdamRJensen can you plot the data from yesterday and see if we are ok with how the measurements look? Then I suggest we close this.

AdamRJensen, 2025-05-02:
Here's the measurement of the SR300

[image]

In fact they all look good.

closed 2025-05-02 by AdamRJensen