AdamRJensen/SARC #70: Does MS80SH not have a heater current we can log?

AdamRJensen, 2025-05-13:
[image]

AdamRJensen, 2025-05-14:
> It sounds like the heater is on regardless of the ambient temperature, so I guess if we are interested, we could log registers 28+29 to identify cases where, for some reason, the heater might not have been in operation. Not ideal, but I don't think that anything else can be done.

Is this register adjacent to registers that we are currently logging? I.e., can we get this for free or do we need to make an additional modbus call?

IoannisSifnaios, 2025-05-15:
> Is this register adjacent to registers that we are currently logging? I.e., can we get this for free or do we need to make an additional modbus call?

We can get it in the same call. It will marginally increase the processing time due to the extra parameter. I will implement it.

closed 2025-05-15 by IoannisSifnaios