AdamRJensen/SARC #73: EKO MS80SHplus

AdamRJensen, 2025-05-20:
The DNI returned by the EKO MS80SHplus is not accurate.

Currently, we are logging the corrected DHI and DNI (registers 10 and 11).
[image]

However, the instrument doesn't seem to be taking the correction measurements (it only stops at the middle of the sensor). This is probably causing an incorrect adjustment to be made. We are in dialog with Mario about this.

AdamRJensen, 2025-05-22:
[Full register map](https://eko-instruments.com/us/wp-content/uploads/sites/15/2025/05/MS-80SH-Plus-Mod-Bus-Register-Map.pdf)

AdamRJensen, 2025-05-22:
Consider logging the band offset: registers 13 and 15

[image]

AdamRJensen, 2025-05-22:
@IoannisSifnaios Changed the EKO MS80SH+ band offset to 10 degrees (it was erroneously set to zero degrees before). Approximate time was 16.20 UTC time on 22-05-2025.

IoannisSifnaios, 2025-05-23:
On 23-05-2025 at 13:00, I updated the secondary script to include the measurements from Registers 13 and 15. I ended up bundling all registers from 7 to 20 in one call (I had to increase the timeout a bit). Unfortunately, data from this morning was lost due to the change of structure in the saved parameters.

@AdamRJensen, make sure to check if the saved parameters make sense, also from when we included the uncorrected measurements (possibly the 20th of May?). It is possible that things have been saved under a wrong name.

AdamRJensen, 2025-05-27:
Measurements from MS80SHplus seemed to have stopped on the 25th of May. The recently added ancillary measurements (band_offset) seemed to be logged correctly.
[image]

Based on our talk today with Justin, we might consider reducing the sampling frequency of the main MS80SHplus scan, e.g., to 2 seconds?

AdamRJensen, 2025-06-03:
> If you want to change the scan frequency, then we will have to put it in a SlowScan (there can be only one main scan). Then you risk that it might not measure all the time.

I think this is an issue caused by trying to log it too frequently - Justin was mentioned that he had run into the same thing when trying to sample sub-second. We are essentially sub-second given that we are making 3 different modbus calls each second.

It would be great if you could unplug it and put it on the slow scan (I would accept 10 second measurements I suppose).

IoannisSifnaios, 2025-06-04:
> I think this is an issue caused by trying to log it too frequently - Justin was mentioned that he had run into the same thing when trying to sample sub-second. We are essentially sub-second given [...]

Our Slow_table is logged every 15 sec. If we sample the MS80SH+ every 10 sec, then it might be a bit uncertain what values will be stored in the table. Two possible fixes:
1. We log the MS80SH+ every 15 sec (easiest solution)
2. We log it every 10 sec, and we create and export a new slowtable that has only this instrument

Thoughts?

AdamRJensen, 2025-07-10:
The serial number of the new MS80SHplus is S24067.011

The new unit was put into operation at 13:32 (15:33 local time) on 2025-07-10.

AdamRJensen, 2025-07-11:
Notes from installing the new unit on 2025-07-10
> The new unit worked straight out of the box with the original wires. To try and figure out what was wrong, we connected the new control unit to the old motor, which worked fine. Then we swapped out the old control unit and used it with the same connections and the new motor, but this did not work. So it definitely seems to be an issue with the control unit.

IoannisSifnaios, 2025-07-11:
The script has been updated with the new serial number. It needs to be send to the datalogger (see [here](https://github.com/AdamRJensen/SARC-secret/pull/2)).

AdamRJensen, 2025-07-11:
GHI Register 7 vs 17
Response from EKO:
>Register 7 will output the GHI per measurement cycle of the RSB. Register 17 is the direct reading of the pyranometer every second, so this may include periods where the pyranometer is unshaded but also shaded or partially shaded by the band. It is only intended for diagnostics, like checking for night time offsets.

My comment:
> I would just expect then the upper boundary of the GHI reading (register 17) to correspond to the GHI (register 7).

[image]

AdamRJensen, 2025-07-14:
- [ ] Change parameter names

IoannisSifnaios, 2025-07-17:
The new unit is logging fine. Parameter names have been updated.

[image]

closed 2025-07-17 by IoannisSifnaios