AdamRJensen/SARC #82: Log individual Solarband channels

AdamRJensen, 2025-06-13:


IoannisSifnaios, 2025-06-19:
It does not look like you can register the voltage from both GHI sensors. We do log the voltage for GHI (I guess the average of the two?) and DHI.

These are the parameters we log:
[image]

And this is the full mobdus map:
https://www.spectrafy.com/sites/default/files/2025-03/SolarBand-C3%20ModBus%20Register%20Map.pdf

AdamRJensen, 2025-06-19:
> But do you see any value in logging the two individual GHI sensors?

Yes I do. I think it would be nice to log all four individual sensors. I believe that we can get this in one call without slowing things down?

IoannisSifnaios, 2025-06-19:
> > But do you see any value in logging the two individual GHI sensors?
> 
> Yes I do. I think it would be nice to log all four individual sensors. I believe that we can get this in one call without slowing things down?

I mean we now ask until register 27 and these are in at 50, so there will be a bit of a delay if we put them in the same call (but I think there is no limit in how many registers we ask for)

AdamRJensen, 2025-07-07:
Started logging the DHI_CH2 on 2025-07-07 around 15:50 UTC (17:50 local time).

closed 2025-07-09 by AdamRJensen