AdamRJensen/SARC #91: Licor leveling issue

AdamRJensen, 2025-07-04:
The Licor bubble level is shot. Admittedly this was a very old leveling plate. 

[image]

AdamRJensen, 2025-07-11:
# Licor bubble level was shut
df.loc['2025-06-30 13:25': '2025-07-10 13:32', [c for c in df.columns if c.startswith('Licor')]] = np.nan

The end time of the logbook was updated.

AdamRJensen, 2026-09-10:
The licor calibration constant is 100, i.e., you multiply the mV signal by 100 to get W/m2.

IoannisSifnaios, 2026-11-13:
Although there was no justification in this issue, the suggested period looks reasonable based on this plot:

[image]

AdamRJensen, 2026-11-14:
> The Licor leveling plate was changed at 13:32 (15:32 local time) on 2025-07-10. The leveling plate used was another old one, while we wait for Licor to ship a new leveling plate.

@IoannisSifnaios I don't see this entry in the logbook for some reason...

AdamRJensen, 2026-11-14:
The period did not seem suspicious from the heatmaps, but I agree that the plot above seems to justify the period.

[image]