AdamRJensen/SARC #100: Infrequent logging of Razon

AdamRJensen, 2025-07-14:
[image]

At 14:06 (16.06 local time) on 2025-07-14 the timeout was increase from 0.1 seconds to 0.15 seconds.

AdamRJensen, 2025-07-15:
> On 2025-07-15, the Razon was set to log every 30 seconds.

Let's not close this before we know that it works

IoannisSifnaios, 2025-07-17:
On 17-07-2025 13:20 local time, I removed the if-statement from the Razon:

old code:
```python
If SecondsIntoMinute = 0
ModbusClient(Razon_190116_result_code,ComC3,19200,Razon_190116_modbus_addr,4,Razon_190116_data(),118+1,5,MODBUS_RETRIES,MODBUS_TIMEOUT,2)
      If Razon_190116_result_code <> 0 Then Razon_190116_data() = NAN
EndIf
```

new code:
```python
ModbusClient(Razon_190116_result_code,ComC3,19200,Razon_190116_modbus_addr,4,Razon_190116_data(),118+1,5,MODBUS_RETRIES,MODBUS_TIMEOUT,2)
If Razon_190116_result_code <> 0 Then Razon_190116_data() = NAN
```

This actually works better, missing only a couple of seconds here and there (at 1-sec sampling).

Note that since Razon measures at a minute interval, the logged values change every minute. However, there should be multiple measurements per minute now.

AdamRJensen, 2025-07-19:
Still not quite solved, but a great improvement:

[image]

```python
razon = [c for c in df.columns if c[:2] in ['PH', 'PR', 'Ra']]
df[razon].isna().resample('1h').mean().plot()
```

closed 2026-10-30 by AdamRJensen