AdamRJensen/SARC #103: No logging of MS80plus_S24067011_rh_per

AdamRJensen, 2025-07-29:
The column ``MS80plus_S24067011_rh_per`` is all nans the past two weeks (ending on 2025-07-28).

@IoannisSifnaios could you check up on this? I would imagine that something is wrong in the script.

IoannisSifnaios, 2025-07-30:
The reason is that we simply do not log this parameter! :-)

It is an unsigned integer while all other params are signed:

[image]

So in order to save one Modbus call, we removed it (see also in #73 at the bottom which parameters are logged in a plot - and RH is not one of them).

AdamRJensen, 2025-08-11:
Removed from the datalogger script on 2025-08-11.

closed 2025-08-11 by AdamRJensen

AdamRJensen, 2026-10-27:
Added to columns to discard in https://github.com/AdamRJensen/SARC/pull/190/changes/e44f74b18f807bc90fdca95578bac0b21a9dedf4.